etf-unibl/sava-drina #24: Implement CODEC parametrization module on FPGA

JovanaAre, 2023-02-02:
Dati VHDL opis modula za parametrizaciju CODEC-a WM8731.

Potrebno je konfigurisati CODEC da radi u loopback režimu.

JovanaAre, 2023-02-02:
Procijenjeno vrijeme za izradu zadatka je 15h.

JovanaAre, 2023-02-03:
Aktivnosti sprovedene na izradi zadatka:

- Istraživanje loopback mehanizma.
- Upoznavanje sa kreiranjem ROM komponente u Quartus softveru.
- Upoznavanje sa kreiranjem .mif fajla (fajla za inicijalizaciju memorije) u Quartus softveru.
- Kreiranje 1-port ROM memorije sa inicijalizacionim fajlom.
- Kreiranje 2 PLL komonente za potrebe implementacije loopback-a.

LTamara1013, 2023-02-10:
Vrijeme je utrošeno na :
-izučavanje loopback mehanizma
- spuštanje programa na ploču i omogućavanje ulaza i izlaza

JovanaAre, 2023-02-12:
Ispoštovan je _Coding Style_ za sve VHD fajlove koji nisu automatski generisani. Ovo ograničenje uključuje i nazive portova komponenata u VHD fajlovima koji uključuju njihovo instanciranje, tj. nije bilo moguće ispoštovati konvenciju __i_ za ulazne i __o_ za izlazne portove automatski generisanih komponenti , jer su nazivi portova generisani bez tih sufiksa,